rubenbe/opensoho #3: Initial v0.2.0 feedback

smithbill17, 2025-08-04:
1. Can we have a version number displayed in the top left under the logo image?
2. Can there be a link that shows a full change log in the app without having to go to GitHub & view Releases?
3. The logo image on my phone is a white circle. Can we have something more like this which suggests a unified network of routers: https://pngtree.com/so/network-logo
4. Can you redesignate SOHO from "Small Office Home Office" which says little about what this does to "Small OpenWRT Home & Office" which implies it's for small OpenWRT based networks.
5. It's not very intuitive to click the Collections breadcrumb to show/hide the menu. It would be more logical to click the 3 PocketBase icons on the left
6. Can we get rid of the 'System' collections - only "_superusers" & "_authOrigins" seems to contain anything 
7. Can you get rid of "id" column in EVERY collection, not just Clients. Can you hide/remove it completely rather than just move it to the last column
8. Can you rename "connected_clients" to be "Clients" and then remove the existing "Clients" collection. Instead, there should be a collection which is the DIFFERENCE between the current Clients & Connected_clients which gives the "Offline Clients" (ie. every previously seen client which isn't currently connected)
9. Can you retrieve client IP address & hostname? Either IP address or hostname should be column 1 & MAC address column 2 or 3
10. The Collection menu order should be fixed as: Devices (ie. routers & access points), Radios, Clients (ie. connected_clients), Offline (ie. current clients - connected_clients)
11. Radios collection should include SSID/WiFi fields (or perhaps include a new collection which is SSID details)
12. Can we include 'Channel' as a field?
13. Is it possible to split the screen horizontally so Devices appears in the top half, then selecting any one Device record updates the bottom pane with a collection of Clients connected to the selected Device
14. Really need a chart of signal levels for each client

smithbill17, 2025-08-04:
The OpenWISP packages I installed on my OpenWRT routers presumably are designed for OpenWISP? What else can they do & what more info can be grabbed from them? Can they also be used to 'control' the OpenWRT devices? eg. radio enable/disable? reboot? disconnect client? etc

smithbill17, 2025-08-04:
With OpenWRT, is it possible to 'ban' a client (ie. by MAC address) from connecting to any given OpenWRT devices/SSID? Could OpenSOHO implement a function to 'ban' connected_clients from its currently connected device, thereby effectively forcing it to only connect to some other SSID or device??

smithbill17, 2025-08-04:
Could functionality be expanded to include 'alerts' for clients or devices like 'when Phone client changes to a different Device > send an alert'  (probably this is straying into OpenWISP functionality?) ?

rubenbe, 2025-08-05:
* Can we have a version number displayed in the top left under the logo image?
  * I've hidden the pocketbase version now, but I'll show the opensoho version instead in the bottom right corner.
* Can there be a link that shows a full change log in the app without having to go to GitHub & view Releases?
  * There will be a link to the github releases page where the changelog can be found.
* The logo image on my phone is a white circle. Can we have something more like this which suggests a unified network of routers: https://pngtree.com/so/network-logo
  * Do you only see the white circle? The text should be there. I gave it a quick try before, but there was nothing that was visually appealing enough. Anyway a logo is not my priority.
* Can you redesignate SOHO from "Small Office Home Office" which says little about what this does to "Small OpenWRT Home & Office" which implies it's for small OpenWRT based networks.
  * i do like the suggestion, maybe "Setup OpenWRT for Home and Office"
* It's not very intuitive to click the Collections breadcrumb to show/hide the menu. It would be more logical to click the 3 PocketBase icons on the left
  * That was my plan too, but it's harder to implement, so the collections it was on a Sunday evening. Although I'll change it into an icon, which makes it more discoverable.
* Can we get rid of the 'System' collections - only "_superusers" & "_authOrigins" seems to contain anything
   * Next version
* Can you get rid of "id" column in EVERY collection, not just Clients. Can you hide/remove it completely rather than just move it to the last column
   * Not my biggest priority, but it is indeed an implementation detail that can be hidden.
* Can you rename "connected_clients" to be "Clients" and then remove the existing "Clients" collection. Instead, there should be a collection which is the DIFFERENCE between the current Clients & Connected_clients which gives the "Offline Clients" (ie. every previously seen client which isn't currently connected is therefore an 'offline' client).
  * The offline clients list is indeed a good idea, it could be
* Can you retrieve client IP address & hostname? Either IP address or hostname should be column 1 & MAC address column 2 or 3. The OpenWISP daemon does not send hostnames, so that won't arrive soon. It seems that it sends IP addresses, but not as part the associated clients, so needs more investigation.
* The Collection menu order should be fixed as: Devices (ie. routers & access points), Radios, Clients (ie. connected_clients), Offline (ie. current clients - connected_clients). I'm not sure if OpenWRT 'Interfaces' should be included as a collection (not sure what details they can provide - DHCP perhaps??).
I agree, but not a high priority now.
* Radios collection should include SSID/WiFi fields (or perhaps include a new collection which is SSID details)

* Can we include 'Channel' as a field for Radios (preferably the actual channel & not just 'auto')?
Could be useful indeed.
* Is it possible to split the screen horizontally so Devices appears in the top half, then selecting any one Device record updates the bottom pane with a collection of Clients connected to the selected Device (selecting no device shows either ALL connected clients or no clients until you select one or more devices).
No splitscreens these are tricky to implement, but clicking through from the devices to a filtered list of connected clients should be easier.
* Really need a chart of signal levels for each client
I agree, but not the highest priority, I first want to have more settings that can be configured.

smithbill17, 2025-08-05:
I've read through your updates on my comments & agree with pretty much everything, although I would like to see the priority on some of the 'cosmetic/useability' aspects placed above the functional possibilities.

To me, the unique thing about OpenSOHO is giving the user a *unified* view across all their devices. I don't really see users wanting to control or modify their router settings from OpenSOHO (beyond say rebooting devices) as effectively that's trying to replace Luci for router management and is likely to lead to things getting screwed up if it fails to apply changes across-the-board.

rubenbe, 2025-08-05:
With our discussions/views it's indeed clear that there are two distict usecases: monitoring and configuring which is currently not reflected in the UI. I'll see if there is an easy method of separating them (similar as with the system/pinned subsections in the (currently called collections).

With regard to the configuration, that's my main goal (and also largely openWISPs) is to not have to log in Luci on every single device to click around configuring something (certainly leading to inconsistencies). To prevent the biggest screw-ups, the OpenWISP daemon on each AP will actually check if it can reach the central server after applying the UCI config and if not rollback to the previous version. Configurations I'm thinking about is adding VLANs on each device, which is something I don't want to have to do clicking around on each of the APs. Also I'm heavily investing my (spare :wink: ) time into unit testing as much as possible, since I'm dogfooding this while developing and I don't want to break my own network either :D
Also it is not my intention to replace Luci as a config tool whatsoever (too complex, too much work, unrealistic) But only the parts that need configuration on multiple devices (so SSIDs & passwords, VLANs)

I see where your focus on UI polishing comes from now ;)
Can you tell which graphs you are thinking about? Some of them might be easy to add.
Although one plan was to add an API for that, and allow the graphing by other better tools (e.g. Home Assistant/ Grafana)

rubenbe, 2025-08-06:
> I think some kind of simple native chart feature in OpenSOHO for signal levels would be good. For anything more complicated, then I guess an API to allow HA/Grafana to be used, is probably best.

I'll have a look for the signal levels, since pocketbase comes with a basic graph for the API call frequency, this might be quite straightforward.

smithbill17, 2025-08-08:
I've noticed that the Radios collection doesn't actually reflect which radios are currently enabled.

I've recently wired one of my Access Points to the main Router and purposely disabled the 5GHz 'backhaul' wireless radio, but it's still showing in the radios collection. There's no field/column that indicates if a radio is enabled or not.

smithbill17, 2025-08-10:
It seems to me, that OpenSOHO keeps setting my OpenWRT devices radios to 'fixed' bands & I keep having to log into Luci on each device & edit each radio to set them back to 'auto'.

I'm not sure how or why it's doing this (at least I'm guessing it's OpenSOHO using the OpenWISP packages on the devices to control my radio channels).

OpenSOHO doesn't currently show the radio channels anywhere, so it's difficult to know which devices have changed & when. And I don't know what rationale OpenSOHO would be using to determine which channel to select.

rubenbe, 2025-08-10:
> So if "disabled" in OpenSOHO, it'll stop changing the config of the Radios, but it'll still read all the parameters, so if I log into Luci and make a change, it'll be reflected in OpenSOHO?
Yes, indeed
> How is it choosing a frequency/channel? What criteria/algorithm is OpenSOHO using to choose? I have 9 Access Points, so either I let them all do their own thing with channels, or I have to work out what devices & AP's are where and try to choose appropriately.

It chooses the lowest "free" channel (see the openwrt docs for mor information). OpenSOHO does not do smart configuration. It will only take the current configuration of the Radio and make it fixed. So yes, if you manually want to configure things, there is some work to do (depending on how big your network is, running one channel analysis via luci is sufficient)

rubenbe, 2025-08-10:
> I've noticed that the Radios collection doesn't actually reflect which radios are currently enabled.

This should be fixed in the new 0.3.0 release
Also added the DHCP leases. Although it is not yet integrated in the clients list
And some cosmetic changes in the UI.