rubenbe/opensoho #8: Enhancement: expand wireless configuration options for roaming

D43m0n, 2025-08-28:
Currently OpenSOHO is able to manage the following `wifi-iface` configuration options:
```
        option bss_transition '1'
        option device 'radio0'
        option disabled '0'
        option encryption 'psk2+ccmp'
        option ft_over_ds '0'
        option ft_psk_generate_local '1'
        option ieee80211r '1'
        option ieee80211v '1'
        option key 'XXXXXXX'
        option mode 'ap'
        option network 'lan'
        option ssid 'OpenSOHO-managed'
```
Is it possible to expand this to add addition options to further enhance roaming options, for example:

**802.11r (Fast BSS transition)**:
```
    option reassociation_deadline '65535'
```
**802.11v (BSS transition management frames)**:
```
    option time_advertisement '2'
    option time_zone 'CET-1CEST,M3.5.0,M10.5.0/3'
    option wnm_sleep_mode '1'
```
**802.11k (Neighbor reports)**:
```
    option ieee80211k '1'
```

D43m0n, 2025-08-29:
> I'll probably expose the human readable version to the user: `Europe/Amsterdam` or equivalent.
Let's try that. The OpenWRT is not very specific about that. In LuCI I can select a prefilled value `CET-1CEST,M3.5.0,M10.5.0/3`. I didn't enter (type or copy) this myself. In LuCI I entered the timezone in that human readable version in LuCI -> System -> System -> System properties -> Timezone. I'm guessing OpenWRT has an internal lookup table that makes this translation?

In `/etc/config/system` that other notation is available although I didn't enter it:

```
config system 'system'
        option conloglevel '8'
        option cronloglevel '7'
        option hostname 'OpenWRT'
        option log_proto 'udp'
        option log_size '128'
        option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option ttylogin '0'
        option urandom_seed '0'
        option zonename 'Europe/Amsterdam'
```

rubenbe, 2025-08-31:
Have a look at the v0.6.0 release, it should implement these configuration flags.