rubenbe/opensoho #10: Error unauthorized when pulling docker image from ghcr.io

D43m0n, 2025-09-03:
When I pull the (first) docker image from ghcr.io, I get an `unauthorized` error message:

```
Pulling opensoho (ghcr.io/opensoho/opensoho-da212a3daf23f854555a4afaf75d4a02:v0.6.0)...
ERROR: Head "https://ghcr.io/v2/opensoho/opensoho-da212a3daf23f854555a4afaf75d4a02/manifests/v0.6.0": unauthorized
```

I use `docker-compose` to maintain my docker images/containers. I also pull the Home Assistant image from ghcr.io without authentication. Could it be a public/private flag on this registry?

majkrzak, 2025-09-09:
On the package view, you should be able to get into the "Package settings" on the bottom of right stack.
There, at the end, in the danger zone, you can find "change visibility" setting.

rubenbe, 2025-09-09:
Thanks a lot @majkrzak, that was indeed the magic button that I was looking for (very well hidden), this should be fixed now!