rubenbe/opensoho #15: invalid login credentials after updating to v0.7.2

D43m0n, 2025-09-29:
I'm running OpenSOHO in a docker container. I've updated to v0.7.2 a few days ago and am unable to login the web interface. I get the message: `Invalid login credentials` in the web interface. When I enable more logging with the `--dev` flag, I see:
```
 [0.00ms] SELECT `_superusers`.* FROM `_superusers` WHERE `email`='REDACTED' LIMIT 1
 [0.00ms] SELECT `_superusers`.* FROM `_superusers` WHERE `email`='REDACTED' LIMIT 1
 ERROR POST /api/collections/_superusers/auth-with-password 
 └─ Failed to authenticate.
    └─ invalid login credentials
```

To solve this, I updated the password inside the container using `./opensoho superuser update <email> <new password>`. I get feedback that this succeeded. After that I still can't login.

Docker-compose:
```
  opensoho:
    image: ghcr.io/opensoho/opensoho:latest
    container_name: opensoho
    mem_swappiness: 0
    command: serve --http 0.0.0.0:8090 --dev
    environment:
      - OPENSOHO_SHARED_SECRET=${OPENSOHO_SHARED_SECRET}
    volumes:
      - "${CONFIG_DIRECTORY}/opensoho/pb_data:/ko-app/pb_data"
    ports:
      - 8090:8090
    restart: unless-stopped
```

Although issue title says "v0.7.2", I'm not entirely sure that version is indeed included in the container image:
```
/ko-app # ./opensoho -v
opensoho version (untracked)
/ko-app # ./opensoho --version
opensoho version (untracked)
```

D43m0n, 2025-09-30:
Can confirm that works:

```
  opensoho:
    image: ghcr.io/opensoho/opensoho:latest
    container_name: opensoho
    mem_swappiness: 0
      #command: serve --http 0.0.0.0:8090
    command: serve --http 0.0.0.0:8090 --dev
    environment:
      - OPENSOHO_SHARED_SECRET=${OPENSOHO_SHARED_SECRET}
    working_dir: /ko-app
    volumes:
      - "${CONFIG_DIRECTORY}/opensoho/pb_data:/ko-app/pb_data"
    ports:
      - 8090:8090
    restart: unless-stopped
```
When using the above docker-compose config I can login with both accounts.

I'm hesitant to close this issue since I haven't had to explicitly set the `working_dir` attribute before in any of my container images. Also because #13 is not closed (yet).

rubenbe, 2025-10-01:
Yes, please do keep it open, since I want to have a better look a the ko options, to set the default working directory in the docker container

D43m0n, 2025-10-05:
It indeed fixes my issue! Now running with:

```
  opensoho:                                                                                                                                                                                                                                                                                       
    image: ghcr.io/opensoho/opensoho:latest                                                                                                                                                                                                                                                                                            
    container_name: opensoho                                                                                                                                                               
    mem_swappiness: 0                                                                                                                                                                      
    command: serve --http 0.0.0.0:8090                                                                                                                                                                                                                                                                               
    environment:                                                                                                                                                                           
      - OPENSOHO_SHARED_SECRET=${OPENSOHO_SHARED_SECRET}                                                                                                                                                                                                                                                                                             
    volumes:                                                                                                                                                                               
      - "${CONFIG_DIRECTORY}/opensoho/pb_data:/ko-app/pb_data"                                                                                                                                                                                                                                              
    ports:                                                                                                                                                                                 
      - 8090:8090                                                                                                                                                                          
    restart: unless-stopped   
```

closed 2025-10-05 by D43m0n